Nic30/hwt #21: Error with sensitivity lists in If statement

benreynwar, 2019-08-16:
I'm getting the error
```
File "/home/ben/Code/hwt/hwt/hdl/ifContainter.py", line 193, in _discover_sensitivity                                                
    assert self._sensitivity is None, self
AssertionError: If(a._eq(0x1),
    intermed_next(0x1),
)
Uncaught exception. Entering post mortem debugging
Running 'cont' or 'step' will restart the program
> /home/ben/Code/hwt/hwt/hdl/ifContainter.py(193)_discover_sensitivity()                                                               
-> assert self._sensitivity is None, self
```
when running the following code:
```
from hwt.synthesizer.unit import Unit
from hwt.hdl.types.enum import HEnum
from hwt.hdl.types.bits import Bits
from hwt.code import If
from hwt.interfaces.std import Signal, VectSignal
from hwt.serializer.verilog.serializer import VerilogSerializer

class Minimal(Unit):

    def _declr(self):
        self.clk = Signal()
        self.a = VectSignal(2)
        self.b = VectSignal(2)
        self.c = Signal()._m()

    def _impl(self):
        one = self._sig('one', Bits(1))
        intermed = self._reg('intermed', Bits(1))
        one(1)
        self.c(intermed)
        If(one._eq(1), (
                If(self.a._eq(1), (
                        intermed(1),
                        )),
                If(self.b._eq(1), (
                        intermed(0),
                        )),
                If(self.b._eq(2), (
                        intermed(1),
                        )),
            ))
''

def main():
    from hwt.synthesizer.utils import toRtl
    u = Minimal()
    contents = toRtl(u, serializer=VerilogSerializer)
    filename = 'minimal.v'
    with open(filename, 'w') as f:
        f.write(contents)

if __name__ == '__main__':
    main()

```

Nic30, 2019-08-17:
Hello, error tells that the statement was discovered twice while analysing the parent statement, this can happen if there are two references on same objects.

However from just brief look I do not see such a thing there so it really looks as some kind of bug.
Maybe the constant value `one` is somehow wrongly interpreted  and this results in to this.
I will look in to it probably tomorrow.

Nic30, 2019-11-07:
The problem was actually optimiser of the statements which did not understand why the sub-statements are driving same signal and why it was not possible to merge such a statements to a one.
Fixed in c1f3506a6a20f073db0a9e63f176ff5d63d66436, tested in https://github.com/Nic30/hwtLib/blob/master/hwtLib/examples/statements/constCondition.py

closed 2019-11-07 by Nic30