Nic30/hwt #34: IP core importing

Nic30, 2020-02-19:
XCI contains parameters of ipcore.
component.xml contains definitions of bus interfaces.

Import xci as a Unit instance. 
* component.xml with pregenerated values usually stored in <proj>.srcs/sources_1/bd/top_name/ip/<ip_name> together with xci

lionheart117, 2020-06-13:
The xci is a format come from Xilinx, isn't it?

Nic30, 2020-06-14:
The one of formats used by Xilinx Vivado is IP-XACT (usually described using component.xml) from this it is possible to generate configured ip core (.xci file generated by vivado after ipcore is added to a project).

Nic30, 2020-12-16:
Example files generated in ipcore folder
[mem_axi_register_slice.zip](https://github.com/Nic30/hwt/files/5701597/mem_axi_register_slice.zip)