Nic30/hwt #40: Issues with basic example

kiteloopdesign, 2021-06-11:
I am getting

`No module named 'hdlConvertorAst.translate._verilog_to_basic_hdl_sim_model'`

when doing `python simple.py`

I have installed both `hwt` and `hwtlib` packages, and I see how all dependances got installed properly. If I do a 

`pip freeze | grep hdlConvertorAst`, I get `hdlConvertorAst==0.8`

Any idea on how to debug this further?

Thanks!

Nic30, 2021-06-11:
Hello this is because I did not release a new package of the hwt yet but I did release a new version of hdlConvertorAst already.

Please use:
```
pip3 install --upgrade --force-reinstall --no-cache-dir -r https://raw.githubusercontent.com/Nic30/hwt/master/doc/requirements.txt git+git://github.com/Nic30/hwt.git@master#egg=hwt
```
I need to write some verifications before new release of hwt and hwtLib it should take a day, but currently there is a little time.
~week.

Now I just checked in balk virtualenv and it should work. But do execute it using module path rather than file path (`python simple.py` -> `python3 -m hwtLib.examples.simple`) in order to avoid problems with PYTHONPATH.
```
cd hwtLib
$ python3 -m hwtLib.examples.simple
```
```
LIBRARY IEEE;
USE IEEE.std_logic_1164.ALL;
USE IEEE.numeric_std.ALL;
--
--    In order to create a new unit you have to make new class derived from Unit.
--
--    You can use sphinx-hwt plugin for sphinx document generator
--    to generate interactive schematic and documentation.
--    Schematic is generated by directive bellow.
--
--    .. hwt-autodoc::
--    
ENTITY SimpleUnit IS
    PORT(
        a : IN STD_LOGIC;
        b : OUT STD_LOGIC
    );
END ENTITY;

ARCHITECTURE rtl OF SimpleUnit IS
BEGIN
    b <= a;
END ARCHITECTURE;
```

Nic30, 2021-07-02:
Hello,
all mentioned problems should now be fixed, new pypi packages are released:

https://pypi.org/project/hdlConvertorAst/
https://pypi.org/project/hwt/
https://pypi.org/project/hwtLib/

kiteloopdesign, 2021-07-02:
awesome, thanks Nic30!

closed 2021-07-02 by kiteloopdesign