Nic30/hwt #41: add_param_asserts breaks examples

philipaxer, 2022-12-13:
Hi, I am playing with the examples from the wiki. The second one (SimpleUnitWithParam) introduces parameters. However it seems that since a commit in 2021 this is somehow broken or the API has changed in a way that the example isnt working.

Instead of generating generic code where the verilog parameter / VHDL generic is used, hwt will just use the constant and emit an assertion if the parameter was indeed set to that constant. That is not very useful.

regards
Philip

Nic30, 2022-12-15:
Hello @philipaxer ,

I did update https://github.com/Nic30/hwt/wiki/Most-simple-examples
It is possible but it is constraining and that is why Params are converted to its values so they do not need any special care
in python expressions.

Do you, as an user, miss an explanation how to have nice generics? It seems to me that for more complicated designs it is more complicated to write only translatable expressions with generics rather than decipher evaluated expressions with them.
But it could be only a local option, that is why I am asking others.

philipaxer, 2022-12-18:
Hi @Nic30,
this is a tricky one. Since hwl has the feature, I expected it to act as a true parameter - However, I see the complexity involved of keeping it a true parameter/generic. Not all possible expressions in a high-level-language/meta-language that hwt supports can be mapped to a vhdl/verilog parameter which brings you into (and also others) into a squeeze.
It seems the tendency of other flows (Chisel, SpinalHDL, Myhdl, migen, ..) is to use verilog/vhdl as a non-human-readable target format. 
I havent really found the silver bullet here either.

regards
Philip

Nic30, 2022-12-19:
Yes, Chisel3, SpinalHDL, Myhdl, migen, nmigen, PyMTL3, any HLS tool and basically every tool of this kind does it in this way.

If there is not issue I am closing this issue.

If you are using HWT or similar tool on some interesting open source project I am always interested.
There is actually lot of things happening bellow the surface of HWT if you are interested.

Nic30

closed 2022-12-19 by Nic30

philipaxer, 2022-12-19:
Coming from an ASIC background, there are some basic requirements that i have and so far none of the available toolkits (chisel, myhdl, spinalhdl, you name it...) worked for me:
* Ideally Python-based (Scala is not very wide-spread and the learning curve is steeper)
* Retain hierarchy (e.g. myhdl flattens the entire design)
* Retain hierarchy and signal names over compilation runs (to not screw up any synthesis/simulation TCL scripts that depend on HDL level names)
* **Decently** readable target HDL so that UVM verification can be used to debug the design. If i trade-off design work with verification work, that doesnt really help. (SpinalHDL generated internal signals for EVERYTHING that is a nightmare. FIRRTL-based stuff is also not very useful here). The issue is that "decent" is a very subjective statement. Also I am assuming people use a traditional verification flow, which is not common in the open source world (because there is no open source UVM capable event driven simulator available)
* Basic RTL linting. I want a toolkit to prevent me to make CDC/latch mistakes

Ideally i would like to have something that would generate the HDL that i would write myself - but I need to admit that this is probably not possible because of the abstraction/HLS-expressiveness.

At this point i have started to write some python to SV toolkit on my own but i am not where i want to be.

Nic30, 2022-12-20:
I agree with you in all points.

* The hierarchy is not just kept but can be also dynamically altered on demand, for example register file can be extracted etc.
* Readability is must have because of sim./verif and constraints. Also the generated code size must be minimal because just simple things like some PCI-e based SmartNIC can generate hundreds of MB if described in something like chisel(vs 130kB). From this reason hwt supports shared component definitions, merges compatible processes to a single one and tries to make formatting visually nice, ...
* not just CDC, latches but also mapping of registers/memories to busses, combinational loops, resource consumption asserts in the tests ...

> started to write some python to SV toolkit 

* That reminds me the first commit of this project in 2015. If possible consider cooperation, there is on a few of us and lot of work to do.

Status in shot:

* HWT is complete, now building vendor neutral HLS extension mainly for packet processing, superscalar and ML.

Status with more context:

HWT is no feature complete used in commercial projects and relatively stable and tested. However it is somehow just a basic tool which will transpil python on input (similar to Chisel, SpinalHDL, nmigen). However as an input is made from customizable AST node like objects, whole thing is highly customizable. Once AST is loaded any transformation/analysis can be applied before generation of target HDL (like in chisel/FIRTL). Hwt deeply cares about hierarchies of components, interfaces and memories. It is also well integrated with UVM, doc gen. and visualization tools.

However a structural/dataflow description which is written in input AST is not sufficient for all applications. Some applications just require algorithmic description which is translated considering scheduling constraints and target tech./arch. specifics.

Consider for example IDS SmartNIC application (e.g. [this](https://www.usenix.org/conference/osdi20/presentation/zhao-zhipeng). First you need to get packet from physical IO, do a TCP reassembling, then parse it, check checkums then classify it by some NN or regex engine and by some hash tables stored in DDR4 with non blocking cache then you have to build an alert message and send it using TCP+TLS, PCI-e.  +- 1 year of work(2021), now if you move for example from VirtexUS+ to Stratix10 FPGA you can discard nearly all of it because everything is different.

I am currently building hwtHls which adds a special HLS process to HWT designs. This process is basically a multi-threaded program. This input code is translated from python bytecode to LLVM, ABC and Z3 where it is optimized with common program opt. then it is translated to hwtHls virtual machine which natively parallel. Then there is an another set of optimization to generate an efficient arch. for input code. This involves pipelining, scheduling, asynchronous segment extraction, mem. port allocations and many more. The output arch. description is then in HWT format. Everything is directed by scheduling and resource constraints.

In a system like this it is easy to write transformation which will translate a packet operations for any interface, superscalar segment extractions, cache coherency protocol handling and many interesting things.

* The mentioned IDS SmartNIC app is +- 5 day job with this extension (vs. 1 year).
* The repo on github is outdated but I do have just 17 issues to finish so I will switch back to github in next week, it is usable but misses the doc.