Nic30/hwt #42: assigment semantics unclear

philipaxer, 2022-12-21:
Hi @Nic30,

I have the following piece of code (please ignreo the Flow stuff for now). HWT complains that valid has multiple drivers. 

```class Demo(Unit):
    def _declr(self):
        addClkRstn(self)
        
        self.a = Flow()._m()
        self.b = Flow()
    
    def _impl(self):
        data = self._reg(name="data", dtype=self.a.data._dtype)
        valid = self._reg(name="valid", dtype=Bits(1))

        valid(valid & ~self.a.fire())

        If(self.b.fire(),
        data(self.a.data),
        valid(True))
        
        self.a.data(data)
        self.a.valid(valid)
```

What I want to describe the the verilog equivalent of 

```
valid_a <= 0;
if(valid_b) begin
valid_a <= 1;
end
```

So essentially I am wondering if I can have a "last assigment wins" policy. I know this example is not so meaningful and you can easily get around it be using else clauses.  This is more a conceptional question. 
Is there some other channel to ask questions?

philipaxer, 2022-12-21:
By looking how HWT works under the hood, i think what I am asking for is not possible. The reason is that HWT is not aware of the python AST and doesnt know which statement is _first_. This makes all assigments equal citizens.

Nic30, 2023-01-01:
As you mentioned, HWT does not work with python AST. Instead it uses python object and the `If` class and others are just regular classes.

If you want to have statements in a code block as you visually see them you must use a `CodeBlock` object.

```Python
from hwt.code import If, CodeBlock
from hwt.interfaces.std import Signal
from hwt.hdl.types.bits import Bits
from hwt.synthesizer.interface import Interface
from hwt.synthesizer.unit import Unit
from hwt.interfaces.utils import addClkRstn

class Flow(Interface):
    def _declr(self):
        self.data = Signal(Bits(32))
        self.valid = Signal()

    def fire(self):
        return self.valid
    
class Example0philipaxer(Unit):
    def _declr(self):
        addClkRstn(self)
        
        self.a = Flow()._m()
        self.b = Flow()

    def _impl(self):
        data = self._reg(name="data", dtype=self.a.data._dtype)
        valid = self._reg(name="valid", dtype=Bits(1))

        CodeBlock(
            valid(valid & ~self.a.fire()),
            If(self.b.fire(),
                data(self.a.data),
                valid(True)
            )
        )
        
        self.a.data(data)
        self.a.valid(valid)
    

if __name__ == "__main__":
    from hwt.synthesizer.utils import to_rtl_str
    u = Example0philipaxer()
    print(to_rtl_str(u))
```

closed 2023-04-16 by Nic30