Nic30/hwt #43: Instantiate VHDL or Verilog IPs as black-boxes

jesseclin, 2024-02-24:
Hi Nic30:

Is it possible to instantiate a VHDL or Verilog IP as a black-box component like [Spinalhdl](https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Structuring/blackbox.html)? Thanks.

Nic30, 2024-02-27:
Hello @jesseclin ,

there are multiple ways how to do it depending on what exactly you need.
* You can create a black box and exclude it from serialization to VHDL 
 see https://github.com/Nic30/hwtLib/blob/master/hwtLib/xilinx/primitive/dsp48e1.py#L19
 You can automatically generate black-box body by inheriting from EmptyUnit https://github.com/Nic30/hwtLib/blob/master/hwtLib/examples/emptyUnitWithSpi.py#L10
* You can automatically load definition/implementation from VHDL/SV using hdlConvertor, this is advanced topic and I do not have any simple example. If you are interested maybe I can write one.

jesseclin, 2024-02-27:
Hi Nic30:

We need to include some mixed-signal behavioral models in our design, and the first way you mentioned should be OK for our purpose.

However, we are still curious about the second suggestion, so please provide an example to illustrate it if possible—many thanks.