Nic30/hwt #46: Incorrect initialization with Verilog serializer

jesseclin, 2024-03-18:
For the following code:

```python
import math
from hwt.synthesizer.param import Param
from hwt.synthesizer.unit import Unit
from hwt.hdl.types.bits import Bits
from hwt.interfaces.std import Signal
from hwt.hdl.types.array import HArray

class LUTSin(Unit):
   
    def _declr(self):
        self.idx  = Signal(Bits(4))
        self.data = Signal(Bits(8))._m()

    def _impl(self):
    
        lut = self._sig(name="lut", dtype=HArray(Bits(8,signed=True),16), 
                                    def_val=[int(math.sin(i*math.pi*2/16)*127+0.5) for i in range(16)])

        self.data(lut[self.idx])

if __name__ == "__main__":

    from hwt.synthesizer.utils import to_rtl_str
    from hwt.serializer.vhdl import Vhdl2008Serializer
    from hwt.serializer.verilog import VerilogSerializer

    print(to_rtl_str(LUTSin(), serializer_cls=Vhdl2008Serializer))
    print(to_rtl_str(LUTSin(), serializer_cls=VerilogSerializer))

```

Output of VHDL serializer (Correct initialization):

```VHDL
LIBRARY IEEE;
USE IEEE.std_logic_1164.ALL;
USE IEEE.numeric_std.ALL;
ENTITY LUTSin IS
    PORT(
        data : OUT STD_LOGIC_VECTOR(7 DOWNTO 0);
        idx : IN STD_LOGIC_VECTOR(3 DOWNTO 0)
    );
END ENTITY;

ARCHITECTURE rtl OF LUTSin IS
    TYPE arr_t_0 IS ARRAY (15 DOWNTO 0) OF SIGNED(7 DOWNTO 0);
    CONSTANT lut : arr_t_0 := (
        SIGNED'(X"00"),
        SIGNED'(X"31"),
        SIGNED'(X"5A"),
        SIGNED'(X"75"),
        SIGNED'(X"7F"),
        SIGNED'(X"75"),
        SIGNED'(X"5A"),
        SIGNED'(X"31"),
        SIGNED'(X"00"),
        SIGNED'(X"-30"),
        SIGNED'(X"-59"),
        SIGNED'(X"-74"),
        SIGNED'(X"-7E"),
        SIGNED'(X"-74"),
        SIGNED'(X"-59"),
        SIGNED'(X"-30"));
BEGIN
    assig_process_data: PROCESS(idx)
        VARIABLE tmpCastExpr_0 : SIGNED(7 DOWNTO 0);
    BEGIN
        tmpCastExpr_0 := lut(TO_INTEGER(UNSIGNED(idx)));
        data <= STD_LOGIC_VECTOR(tmpCastExpr_0);
    END PROCESS;
END ARCHITECTURE;
```

Output of Verilog serializer (Incorrect initialization)

```Verilog
module LUTSin (
    output reg[7:0] data,
    input wire[3:0] idx
);
    reg signed[7:0] lut[0:15];
    always @(idx) begin: assig_process_data
        data = $signed(lut[idx]);
    end

    initial begin
        lut[0] = 0;
        lut[1] = 1;
        lut[2] = 2;
        lut[3] = 3;
        lut[4] = 4;
        lut[5] = 5;
        lut[6] = 6;
        lut[7] = 7;
        lut[8] = 8;
        lut[9] = 9;
        lut[10] = 10;
        lut[11] = 11;
        lut[12] = 12;
        lut[13] = 13;
        lut[14] = 14;
        lut[15] = 15;
    end

endmodule
```

jesseclin, 2024-03-18:
In  https://github.com/Nic30/hwt/blob/master/hwt/serializer/verilog/serializer.py#L56

line

> for i, _v in enumerate(rom.def_val.val): 

should be change to

> for i, _v in enumerate(rom.def_val):

?

Nic30, 2024-03-18:
Thank you for reporting, it should be fixed now.
New hwt/hwtLib release is waiting for some work in hwtHls, but if you need a new release ping me.

jesseclin, 2024-03-18:
No hurry, for there is still a lot of fun stuff for us to explore at your great work. Thanks.